things4u/ESP-1ch-Gateway-v5.0--OLD #1: works perfect on 4 but keep resets on 5

halukmy, 2017-12-05:
i tried lots of but when i upload v 5, after get gateway id and server starts its going to reset nodemcu

what could be problem?

zenghi, 2017-12-05:
SX1276 can't run.

zenghi, 2017-12-05:
check the reset pin.

digitalWrite(pins.rst, LOW);
delayMicroseconds(10000);
digitalWrite(pins.rst, HIGH);
delayMicroseconds(10000);

halukmy, 2017-12-05:
@zenghi sx1278 too? in v5?

im going to check

zenghi, 2017-12-05:
@halukmy  yes,im using sx1278 in V3.3.

halukmy, 2017-12-05:
@zenghi i guess its hardware problem maybe because i just connect DIO1 others are not connected it my fault of mcu

but V4 going good for example connect to ttn also work as a node, but i got a question on this, 

with mqtt i can downlink an application over ttn network i can see paylod and others but i cant see traffic data on gateway also cant downlink to node

hows that possible to downlink a device? is it possible with v4? thanks zenghi

zenghi, 2017-12-06:
@halukmy Upstream data can be seen on the NS, but the downlink data is not processed, I will try to solve.

Sukuri, 2017-12-06:
@halukmy reset pin shouldn't be connected to any pin.

halukmy, 2017-12-06:
@Sukuri i checked again, i guess my nodemcu got problem, because in same connection v4 works perfect but v5 after conected to wifi and mqtt boom again.. anyway i love this repo and these people

hope @zenghi  can handle the downlink via mqtt , the last step of smart cities on his magical hand and  brain

thanks

halukmy, 2017-12-06:
setupOta:: Started
Ready
IP address: 192.168.0.11
setupTime:: Time not set (yet)
WiFi connected
Time: Wednesday 20:45:25
Gateway configuration saved
WWW Server started on port 80

Abort called

ctx: cont 
sp: 3fff2ac0 end: 3fff2d10 offset: 01b0

>>>stack>>>
3fff2c70:  00000042 00002710 3fff1c2c 4020794d  
3fff2c80:  3ffe8410 00002710 00000000 40208ebf  
3fff2c90:  3fff19f0 3ffe8410 3fff1c2c 4020bc1f  
3fff2ca0:  000000fe 00000047 feefeffe feefeffe  
3fff2cb0:  633a6566 66653a66 3a62383a 343a6566  
3fff2cc0:  65000037 32387073 382d3636 34393662  
3fff2cd0:  00000000 00000000 00000000 feefeffe  
3fff2ce0:  feefeffe feefeffe feefeffe 3fff1ce8  
3fff2cf0:  3fffdad0 00000000 3fff1ce0 402169dc  
3fff2d00:  feefeffe feefeffe 3fff1cf0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

halukmy, 2017-12-15:
@zenghi my nodemcu still reset on v5 
My only DI0 connected to esp12e 

may it be problem?

zenghi, 2017-12-15:
[image]
I have not tried to use 5V,DIO0 not linked, will not make the operation anomalies。

platenspeler, 2018-06-02:
Is this solved, or still occurring?

Maarten

halukmy, 2018-06-03:
still same but i just move to ttgo module, needs update keep sending tx :00 like crazy tons of lines and not works well

Matheus-Garbelini, 2018-06-04:
@halukmy Oh my god !!!

halukmy, 2018-06-04:
whats happend? @Matheus-Garbelini

platenspeler, 2018-06-14:
I close this thread since this problem was described in another issue as well. Alls L0x00 prints should have been removed in the latest version of the software.

closed 2018-06-14 by platenspeler