things4u/ESP-1ch-Gateway-v5.0--OLD #2: downlink how you send to nodes from mqtt or ttn?

halukmy, 2017-12-05:
hello all, i just collect data when node send to gateway(esp12e)  via mqtt parsing so its great i also send downlink via mqtt over ttn network and i can follow traffic on application data 

but when i send data on mqtt network via mqtt i see on application traffic area but coulndt see on traffic on gateway layout also no seen on arduino debug console

any tip?

thanks

zenghi, 2017-12-15:
HI halukmy I successfully use the downlink ，you can view PULL REQUESTS.

halukmy, 2017-12-15:
@zenghi thanks so much I been waiting long so you have any demo? 

Now mqtt works our nodemcu gateway? That's brilliant! Thanks much

zenghi, 2017-12-15:
[image]
[image]
[image]
[image]

Use  ESP8266 to do single-channel receive and transmit,terminal part can see ACK,Platform-side can see the datat storage.

halukmy, 2017-12-15:
Like a dream! Thanks much!

What's your lmic source and version?

zenghi, 2017-12-15:
[image]

SX1278 and esp8266，SX1276 I think the same applies
The terminal uses STM32L151 and SX1278
Maybe I did not understand you very well,hope this helps.

halukmy, 2017-12-15:
Pardon me @zenghi 

i mean, what you use for nodes? for example

https://github.com/lyusupov/SoftRF/tree/master/software/firmware/source/libraries/arduino-lmic 
this lib you advices?

and second question,

is it possible message without gateway with secure? did you work any on it?

zenghi, 2017-12-18:
I'm sorry, I see your message now。
I use the following lib
https://github.com/Lora-net/LoRaMac-node

The second question, not related to the work

halukmy, 2017-12-18:
im using www.dorji.com/docs/data/DRF1278F.pdf with lmic 1.5 , is my dorji work with LoraMac-node? i cant see any examples ino in semtech repo:(

zenghi, 2017-12-19:
If possible, I can give you a set of hardware. But the logistics cycle will be longer.

halukmy, 2017-12-19:
@zenghi you serious? that would be awesome!

halukmy, 2017-12-19:
im trying to sx1278 as gateway btw

zenghi, 2017-12-20:
Yes, at present, the complete circuit is being debugged, and once the success will be sent to a device to you, please send your address to my private mailbox.

halukmy, 2017-12-20:
Thanks @zenghi , i just send mail to 309966755@qq.com from haluky@live.com

halukmy, 2018-06-19:
@zenghi hi Zenghi,

do you have Class-C support for this channel for limic ?

schwiderke92, 2019-03-02:
downlink is work?

richard-ubi, 2019-06-24:
I have encountered such a problem, I sent the data to LoRaServer only uplink, downlink error
[image]