things4u/ESP-1ch-Gateway-v5.0--OLD #5: use of struct LoraUp  parameter in sensorPacket messageSent

maurocordioli, 2017-12-19:
when we enable GATEWAYNODE , 
 in sensorPacket /_sensor.ino 
the function messageSent is called with may be , an outdated parameter list (using message and message length) instead of use of struct LoraUp  parameter .... 
somethig like this : ( but I don't kown yet the use of other members of the struc, I have just tried my first program from my new ESP32 pair ) 
` 
//MC start
struct   LoraUp LU;
memcpy( LU.payLoad, message, mlength*sizeof(uint8_t) );
LU.payLength= mlength ;
  
    int buff_index = buildPacket(tmst, buff_up,LU,  true);
// old call int buff_index = buildPacket(tmst, buff_up,(uint8_t *) (message), mlength, true);
 //MC end`

iotlearner0level, 2018-04-05:
Thanks 	maurocordioli for sharing. worked for me!

Great!!

platenspeler, 2018-06-12:
Thanks for finding this bug! I have patched the latest 5.2.1 version.

closed 2018-06-12 by platenspeler