things4u/ESP-1ch-Gateway-v5.0--OLD #6: device keep ressetting after started , on esp12e nodemcu board

halukmy, 2018-01-27:
what can be problem?

wero1414, 2018-02-04:
Same problem here 👎

halukmy, 2018-02-04:
developer looks dead

Ldsrc2008, 2018-02-07:
Hi, guys,
I have the same issue with my DI mini,I loaded the program with arduno, it works fine,no problem..
But next day I plug in to PC and open putty..
 it keep on reseting after this:
"Gateway configuration saved
WWW Server started on port 80
Abort called
...
ESP8266 rst cause:2, boot mode:(3,6)"
And the :80 server is not working...
Interest thing, if I close the putty and open arduino (mine is 1.8.5) with serial monitor, reset it..
it works again, and does not reset, www server is working.
I read from google, rst cause:2 is related to reset pin or power.I have nothing on the reset pin.
would that be a problem on the usb-serial chip problem? 
my Di mini is with a CH340G, it should have some hardware connection with the reset pin for uploading firmware.
Any one has suggestion to dig into the problem?

lagunacomputer, 2018-02-19:
i have found this mostly only happens when i don't have solid or correct wiring from the nodemcu esp8266 to my 1276/RMF95 module.  Check your wiring

halukmy, 2018-02-20:
@lagunacomputer where did you connected your dio01 your pin? and what you tried to mean yes and no?

lagunacomputer, 2018-02-20:
Esp8266 GPIO15 / pin D8  to both DIO0 and DIO1 on the LoRa module

No means not connected, not needed

halukmy, 2018-02-20:
@lagunacomputer you connect them same port with breadboard? and if with that works well, its so good!

can you able to connect any node to this gateway? (example  node is esp8266 + sx127X )

i just tried allways dont know how to

lagunacomputer, 2018-02-20:
yes
I have tried both 868 and now 915mhz gateway and nodes.  Both work.
http://things4u.github.io/HardwareGuide/hardware_guide.html
http://things4u.github.io/HardwareGuide/ESP8266/ESP-Sensor/esp-lora-sensor.html
use the LMIC code for the ESP8266 node

halukmy, 2018-02-20:
@lagunacomputer  i got 433 sx1278 module

i dont think lmic supports it

what you think?

lagunacomputer, 2018-02-20:
I think you can make it work!

http://wiki.dragino.com/index.php?title=Connect_to_TTN#How_To_add_other_frequency_band_such_as_EU433
https://github.com/matthijskooijman/arduino-lmic/issues/98

halukmy, 2018-02-20:
@lagunacomputer  can i get your email?

halukmy, 2018-02-22:
i just plugged d1 d2 d3 to same gate,

still resets after connect wifi

halukmy, 2018-02-22:
@lagunacomputer  can you explain me to connect lorawan?

first install esp-1ch-gateway to esp12+ sx1278

after add it to ttn gateway 

after use lmic,

add an application at ttn, and add session keys to lmic

than it should send to single channel right?

lagunacomputer, 2018-02-22:
Hi these answers can all be found with a few google searches.  First, make
sure your gateway is connected to TNN.
If your node keeps resetting, it is most likely your wiring between ESP8266
and the radio module.  I don't know about the 1278 or 433mhz, i used 1276
and 868/915mhz.  Also, I am using NodeMCU v2 and v3 modules, so your pins
may differ.

try these links, they are what I used:
https://www.youtube.com/watch?v=ZV_ZY-0Q1lo
https://www.youtube.com/watch?v=Ya-QlEaonLU
https://www.thethingsnetwork.org/forum/t/single-channel-gateway-part-3/11546
http://things4u.github.io/HardwareGuide/hardware_guide.html

halukmy, 2018-02-22:
i have already watched these steps and videos,

i just connected to ttn but how send to ttn server uplink or downlink?

i just create application at ttn add keys to lmic and upload it, it says message send but i dont see any update on ttn or debug single chanel gateway

lagunacomputer, 2018-02-22:
goto http://theipofyourgateway and see if its getting messages, also you
should see traffic in the serial monitor of the gateway.
I use a putty and open 2 serial sessions next to each other so I can see
what is happening, one for the gateway, and the other for the node
[image: Inline image 1]

halukmy, 2018-02-22:
the problem is, i can not see any message while sending it :(

because of it i cant create lorawan

im using this

https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaDuplexCallback/LoRaDuplexCallback.ino

but i need to success wan.. ifts not matter can i get your any contact info?

lagunacomputer, 2018-02-22:
I have no idea if that code even works.  Perhaps contact that author for assistance.

lou71d233, 2018-04-11:
_loraModem.ino

	// Low Noise Amplifier used in receiver
    writeRegister(REG_LNA, (uint8_t) LNA_MAX_GAIN);  			// 0x0C, 0x23
	
       //add this in nodemcu (ESP-12E)
	digitalWrite(pins.rst, HIGH);
        delayMicroseconds(10000);
      
    uint8_t version = readRegister(REG_VERSION);				// Read the LoRa chip version id

platenspeler, 2018-06-24:
So what is the status of this issue? Is the module stable now? I cannot test or support every type of board, so hope this one gets resolved by members of the community :-)

closed 2018-08-07 by platenspeler