things4u/ESP-1ch-Gateway-v5.0--OLD #7: Cannot change Spread factor to SF7

jfmateos, 2018-02-03:
Hi,

I have this in gway.h
`#define _SPREADING SF7

#define _CAD 0

#define _STRICT_1CH	1`

But the gateways stats the following:

`Gateway ID: 2C3AE8FFFF225D1F, Listening at SF9 on 868.10 Mhz.
`

I want to use SF7 becouse I'm using a single channel node that uses SF7

jfmateos, 2018-02-04:
I think the problem is I am not using wifimanager, so once uploaded the sketch for the first time, the SF value of gway.h is stored in SPIFF filesystem.

Later, if I change the SF value in gway.h and upload the sketch again, the SPIFF value of SFN is used.

is there and option to format the file system?

platenspeler, 2018-02-11:
In the latest version 5.0.6 the is in the gway.h file. Just set SPIFF_FORMAT to 1 only one time.

Maarten

platenspeler, 2018-02-11:
O, and every time we change the seetings the sopreading factor should be updated as well. When you change the spreading factor manually, make sure that you press BOTTS and RESETS for example in the webserver. It will write the current configuration to SPIFFS as well (which shouldl prevent you from formatting).

closed 2018-03-03 by platenspeler