things4u/ESP-1ch-Gateway-v5.0--OLD #10: My gateway stops working after a few days

jfmateos, 2018-03-21:
Hi, I have a gateway powered through a phone charger.
The gateways works flawlessly for days and then it just disappear from the wifi network.

How could i troubleshoot this issue?
Does the software store any statistical information about the processor status?

platenspeler, 2018-03-26:
Thanks, I think that the latest version (which I'm trying to get on line this week) will solve WiFi issues. It appears that the ESP8266 has a bug when WiFi.begin() is executed. It does not forget the last WiFi access point connected to, but also does not allow reconnecting to that Access Point. The patch does correct this.
So when the gateway for whatever reason looses the connection for a short time it will now reconnect to the last Access Point. The latest version solves that.

platenspeler, 2018-04-01:
Latest version 5.0.8 survives a WiFi gateway error.

closed 2018-04-01 by platenspeler