things4u/ESP-1ch-Gateway-v5.0--OLD #13: Gate shown in ttn console but no traffic.

iotlearner0level, 2018-04-08:
I am stuck with a strange problem. local webpage of the esp gw shows connected. it shows message received from another lora device. debug level 4 shows messages but nothing on the ttn console in the traffic page. also never seen on the ttn application devices page. i've tried GATEWAY=1 still no traffic on ttn console in gateways page!

here is what the webpage shows:

Counter | Pkgs | Pkgs/hr
-- | -- | --
Packages Uplink Total | 3 | 37
Packages Uplink OK | 3
Packages Downlink | 0
SF7 rcvd | 3 | 100 %
SF8 rcvd | 0 | 0 %
SF9 rcvd | 0 | 0 %
SF10 rcvd | 0 | 0 %
SF11 rcvd | 0 | 0 %
SF12 rcvd | 0 | 0 %

Setting | Value | Set
-- | -- | --
CAD | ON | ON | OFF
HOP | OFF | ON | OFF
SF Setting | AUTO
Channel | 0 | - | +
Debug level | 3 | - | +
Usb Debug | 1
Framecounter Internal Sensor | 1342 | RESET
Gateway Node | ON | ON | OFF
WWW Refresh | ON | ON | OFF
Statistics | 0 | RESET
Boots and Resets | 71 | RESET

LoRa Router | eu.thethings.network
-- | --
LoRa Router IP | 52.169.76.255

LoRa Router | eu.thethings.network
-- | --
LoRa Router IP | 52.169.76.255

from the Serial monitor:
SPIFFS loaded success
Assert=Do Asserts
debug=1
readConfig:: Starting
SSID=dlink-8B44
CH=0
SF  =7
FCNT=1330
DEBUG=3
CAD=1
HOP=0
NODE=1
BOOTS=70
RESETS=0
WIFIS=85
VIEWS=228
REFR=1
REENTS=0
NTPETIM=10736
NTPERR=29
NTPS=121

MAC: , len=17
0:1. WiFi connect to: 
.....WLAN reconnected
Local UDP port=1700
Connection successful
Gateway ID: 30AEA4FFFF6EFBE8, Listening at SF7 on 433.18 Mhz.
setupTime:: Time not set (yet)
WiFi connected
Time: Monday 01:58:10
Gateway configuration saved
WWW Server started on port 80
pRSSI: -110 RSSI: -124 SNR: 2 Length: 26 -> 40 AE 11 1 26 80 0 0 1 FF 4C B4 AD FB 61 A2 DB CA 68 43 67 EE 70 7E B 78 
RXPK:: {"rxpk":[{"tmst":25059907,"chan":0,"rfch":0,"freq":433.174987,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":2,"rssi":-110,"size":26,"data":"QK4RASaAAAAB/0y0rfthotvKaENn7nB+C3g="}]}
RXPK:: package length=207
WiFi connected
rpRSSI: -111 RSSI: -126 SNR: 2 Length: 26 -> 40 AE 11 1 26 80 0 0 1 FF 4C B4 AD FB 61 A2 DB CA 68 43 67 EE 70 7E B 78 
RXPK:: {"rxpk":[{"tmst":29338479,"chan":0,"rfch":0,"freq":433.174987,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":2,"rssi":-111,"size":26,"data":"QK4RASaAAAAB/0y0rfthotvKaENn7nB+C3g="}]}
RXPK:: package length=207
WiFi connected
pRSSI: -111 RSSI: -124 SNR: 2 Length: 26 -> 40 AE 11 1 26 80 0 0 1 FF 4C B4 AD FB 61 A2 DB CA 68 43 67 EE 70 7E B 78 
RXPK:: {"rxpk":[{"tmst":37213771,"chan":0,"rfch":0,"freq":433.174987,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":2,"rssi":-111,"size":26,"data":"QK4RASaAAAAB/0y0rfthotvKaENn7nB+C3g="}]}
RXPK:: package length=207
WiFi connected
PULL <WiFi connected
PKT_PULL_DATA request, len=<12> 1:78:D8:2:30:AE:A4:FF:FF:6E:FB:E8:
>
stateMachine:: NO intr
PULL <WiFi connected
PKT_PULL_DATA request, len=<12> 1:68:A7:2:30:AE:A4:FF:FF:6E:FB:E8:
>
stateMachine:: NO intr
STAT <stat update: <229> {"stat":{"time":"2018-04-09 01:59:59 CET","lati":28.53000,"long":77.37999,"alti":8,"rxnb":3,"rxok":3,"rxfw":0,"ackr":0.0,"dwnb":0,"txnb":0,"pfrm":"ESP32","mail":"hellothereyandex@gmail.com","desc":"My ESP32 Gateway"}}
WiFi connected
>
pRSSI: -107 RSSI: -128 SNR: 12 Length: 16 -> 40 AE 11 1 26 0 3C 5 1 C3 D0 60 6 4A D3 D9 
RXPK:: {"rxpk":[{"tmst":120095929,"chan":0,"rfch":0,"freq":433.174987,"stat":1,"modu":"LORA","datr":"SF?BW125","codr":"4/5","lsnr":12,"rssi":-107,"size":16,"data":"QK4RASYAPAUBw9BgBkrT2Q=="}]}
RXPK:: package length=197
WiFi connected
PULL <WiFi connected
PKT_PULL_DATA request, len=<12> 1:FC:D9:2:30:AE:A4:FF:FF:6E:FB:E8:
>
stateMachine:: NO intr
PULL <WiFi connected
PKT_PULL_DATA request, len=<12> 1:7A:96:2:30:AE:A4:FF:FF:6E:FB:E8:
>
stateMachine:: NO intr
STAT <stat update: <229> {"stat":{"time":"2018-04-09 02:01:59 CET","lati":28.53000,"long":77.37999,"alti":8,"rxnb":3,"rxok":3,"rxfw":0,"ackr":0.0,"dwnb":0,"txnb":0,"pfrm":"ESP32","mail":"hellothereyandex@gmail.com","desc":"My ESP32 Gateway"}}
WiFi connected
>
pRSSI: -107 RSSI: -126 SNR: 12 Length: 16 -> 40 AE 11 1 26 0 3D 5 1 CD 93 8 43 AB 2C 8A 
RXPK:: {"rxpk":[{"tmst":240095931,"chan":0,"rfch":0,"freq":433.174987,"stat":1,"modu":"LORA","datr":"SF?BW125","codr":"4/5","lsnr":12,"rssi":-107,"size":16,"data":"QK4RASYAPQUBzZMIQ6ssig=="}]}
RXPK:: package length=197
WiFi connected
PULL <WiFi connected
PKT_PULL_DATA request, len=<12> 1:74:64:2:30:AE:A4:FF:FF:6E:FB:E8:
>
stateMachine:: NO intr
PULL <WiFi connected
PKT_PULL_DATA request, len=<12> 1:C4:AF:2:30:AE:A4:FF:FF:6E:FB:E8:
>
stateMachine:: NO intr
STAT <stat update: <229> {"stat":{"time":"2018-04-09 02:03:59 CET","lati":28.53000,"long":77.37999,"alti":8,"rxnb":3,"rxok":3,"rxfw":0,"ackr":0.0,"dwnb":0,"txnb":0,"pfrm":"ESP32","mail":"hellothereyandex@gmail.com","desc":"My ESP32 Gateway"}}
WiFi connected
>
pRSSI: -107 RSSI: -126 SNR: 12 Length: 16 -> 40 AE 11 1 26 0 3E 5 1 69 9 9E 35 62 F0 A0 
RXPK:: {"rxpk":[{"tmst":360095934,"chan":0,"rfch":0,"freq":433.174987,"stat":1,"modu":"LORA","datr":"SF?BW125","codr":"4/5","lsnr":12,"rssi":-107,"size":16,"data":"QK4RASYAPgUBaQmeNWLwoA=="}]}
RXPK:: package length=197
WiFi connected
PULL <WiFi connected
PKT_PULL_DATA request, len=<12> 1:57:DB:2:30:AE:A4:FF:FF:6E:FB:E8:
>
stateMachine:: NO intr

I've tried wemos d1 mini and esp32 heltec lora board with the kersing fork, still nothing. it is very frustrating!!

iotlearner0level, 2018-04-08:
Also for the wemos d1 mini connected to aithinker ra01 (based on sx1278) i get the following message continuously:

S_RX:: no RXDONE or RXTOUT intr=0

Matheus-Garbelini, 2018-04-08:
Could be something that changed on ttn? While igo a Trier to use this library but it didnt work ether. Didn't Tried again since.

There a open source loraserver avaliable to deploy in linux servers, should be an alternativa maybe :x

iotlearner0level, 2018-04-09:
Could this be related to udp port 1700 being blocked from the ISP side?

iotlearner0level, 2018-04-09:
UPDATE: I temporarily managed to fix it by taking two steps:
(1) Gateway was not showing properly. on the serial monitor, i did not see PUSH_ACK from the lorawan server. i changed it, and now it is working.
(2) TTN server rejects packets if the frequency of the module used by the gateway is different from that configured in the TTN Console gateway settings. If it is not possible to get the "correct" lorwan module, this problem can be temporarily overcome by faking frequency sent in the lorwan packet json string. in the current version of the code this is line 373 in _rxtx. Change it to:  ftoa(868.10,cfreq,6); replace 868.1 to whatever band is configured in the TTN console settings. of course, this should only be a temporary measure. better get a correct frequency module for your country.

just for reference:
* https://www.thethingsnetwork.org/docs/gateways/packet-forwarder/semtech-udp.html
* https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT