things4u/ESP-1ch-Gateway-v5.0--OLD #17: gateway owner

EI2KK, 2018-05-08:
Hi there..

My gateway is online, I can see it on the map but I have no idea how to add it to my console.. 
I can't see how to add Gateway Key from console to sketch.

On the map there is gateway ID only, any other info is unknown.

craigpeacock, 2018-05-09:
You will need to be a little clearer with what your problem is and terminology, if you want help.

Do I assume you are unable to register the gateway on The Things Network (TTN) console (sketch)? 

If this is the case, under Register Gateway in the TTN Console, you need to click the tickbox "I'm using the legacy packet forwarder". This will then allow you to paste the Gateway ID into the Gateway EUI Field.

EI2KK, 2018-05-09:
OK, I registered it as you said, ID is the same but console displays it as 'not connected'. 
Gateway restarted...
I can see gateway on the map with no owner.

closed 2018-05-09 by EI2KK

jlstanus, 2020-04-28:
@EI2KK Finally owner identity, model, placement has been updated?