things4u/ESP-1ch-Gateway-v5.0--OLD #19: fake packets?

EI2KK, 2018-05-10:
My gateway is in remote area surrounded by hills - I do not think there are any nodes around.

Is it possible to recieve fake packets? Last night I got 3 packets, one with CRC error so I assume the other ones have correct CRC..  but they do not appear in console - Received Messages 0.
(I can't test it with my own node yet - waiting for stuff)

In issue #13 opened on Apr 8 iotlearner0level wrote about the frequency - I have chan:0 -  868.10 and plan for gateway in console set to Europe.

Log shows a little bit different (25Hz) frequency (868.099975) - is that OK for server to accept packet?
I can see my gateway online so there is no port/Internet related issue.

rxPkt:: t=23:20:59, f=0, sf=7, a=4A827F71, flags=50, addr=0, len=41
Err RxPkg:: CRC, =23:30:59
rxPkt:: t=01:31:02, f=0, sf=7, a=C951A51E, flags=50, addr=0, len=7
rxPkt:: receivedCount=93
rxPkt:: t=06:06:15, f=0, sf=7, a=FAB4174E, flags=50, addr=0, len=64
Log

{"rxpk":[{"tmst":514016160,"chan":0,"rfch":0,"freq":868.099975,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":-12,"rssi":-104,"size":41,"data":"L3F/gkqI/DCa1i9zbMkIoVfGAzwObEKU/qH4RpVFwaWX6xFFRPh8qZM="}]}
{"rxpk":[{"tmst":4023015136,"chan":0,"rfch":0,"freq":868.099975,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":-10,"rssi":-109,"size":7,"data":"zx6lUcmy6A=="}]}
{"rxpk":[{"tmst":3355261014,"chan":0,"rfch":0,"freq":868.099975,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":-11,"rssi":-104,"size":64,"data":"M04XtPo0utECqvVzrIz58qctGqirl0ZR7WvVV7b1jPAaLx456GQFqDLB8j3DIyreht9N7UirwgKFvR4aej6zZg=="}]}

platenspeler, 2018-05-22:
Hi,

I hope that the RFM95 together with the CRC error checking will only pass message that are OK. If there are CRC problems, please discard that message.
Unfortunately I cannot check whether this is because and unknown node was received, or an error was passed as successful or etc etc.. I hope that the node ID that is passed tells you more about the node.

I have 10 single channels as test machines working and I setup the "TRUSTED" nodes feature in gateway.h so I can check whether some nodes are not mine. I do not see many (there are a few in Apeldoorn but not many).

Please use the latest software version 5.1.1 that I uploaded today, it enables downloading of logfiles in the browser, Please check again.

regards,

Maarten

platenspeler, 2018-06-12:
I am closing this issue, I do not find such fake nodes myself. Please re-open if you have more information.

closed 2018-06-12 by platenspeler