things4u/ESP-1ch-Gateway-v5.0--OLD #20: No RXDONE

Algram, 2018-05-15:
Hello,
once my gateway starts up successfully and I enable debug mode, it gets spammed with the following messages:

`S_RX:: no RXDONE, RXTOUT, HEADER:: I=  --  , F=0, SF=9, E=0, S=RX  , t=69997239`

What do those mean?

platenspeler, 2018-05-22:
This is a condition that should not happen. It means that we are in receiving state but did receive another interrupt than RXDONE (reception complete) or RXTOUT (timeout).

What version of the software are you running? Please upgrade to the latest version and try again. This is the version I have running on all my Wemos D1 mini gateways and the other ones.

Thanks,

Maarten

platenspeler, 2018-05-22:
So please check the board that you are using and compare to ESP gateway.h 
It looks like you have a pin problem.

closed 2018-06-02 by Algram