things4u/ESP-1ch-Gateway-v5.0--OLD #23: ESP12E RFM95 Wiring!

mojgithub, 2018-05-19:
Hi,

there are lots of different wiring on google for ESP12e and RFM95 but none of those work for me.

_loraModem.h "i have _PIN_OUT==1"

uint8_t dio0=15;	// GPIO15 / D8. For the Hallard board shared between DIO0/DIO1/DIO2
	uint8_t dio1=15;	// GPIO15 / D8. Used for CAD, may or not be shared with DIO0
	uint8_t dio2=15;	// GPIO15 / D8. Used for frequency hopping, don't care
	uint8_t ss=16;		// GPIO16 / D0. Select pin connected to GPIO16 / D0
	uint8_t rst=0;		// GPIO 0 / D3. Reset pin not used	
	// MISO 12 / D6
	// MOSI 13 / D7
	// CLK  14 / D5_

So why are MISO,MOSI, CLK pins commented?

Is this correct wiring?

ESP8266	LORA Board
GND	GND
VCC	VCC
GPIO15	NSS
GPIO13	MOSI
GPIO12	MIS0
GPIO14	SCK
GPIO5	DIO0

If not which one is correct and do i need to change something in loraModem.h at _PIN_OUT code(do i need to uncomment MISO, MOSI etc..)?

In serial monitor i get this:
MAC: AA:CC:a6:CC:09:GG, len=17
Host esp8266-YYYY WiFi Connected to XXXXX
skdhfskdhfs%#%#dhfkfgskdfgsdklfgsdhk5435$%%fgsdfsdjfsdjfg#$%#sdjkfgsdjkfgsdjkfsd
Assert=Do Asserts
debug=2
readConfig:: Starting ERROR:: readConfig, file=/gwayConfig.txt does not exist .. Formatting

File "gwayConfig.txt " exist in folder Arduino\examples\libraries\ESP-sc-gway so why i am getting this error?
After WiFi Connected to XXXXX i get some long corrupted characters.
I cant get Gateway ID which is next step to register TTN.

platenspeler, 2018-05-22:
Please have a look at the documentation: things4u.github.io
It contains more information about pin-out etc.

Maarten

platenspeler, 2018-06-02:
By the way: ESP32 support required some changes to be made to the code. Especially the _PIN_OUT definition in ESP-sc-gway.h file. I have added tyhat support in version 5.2.0 for the TTGO board. Please try whether you can get yout board supported as well.

platenspeler, 2018-06-12:
Did you find what you are looking for?
The PIN definition is set in ESP-sc-gway.h file. In _loraModem.h file you can define your own pin-out. What you see in the _PIN_OUT==1 definition for Hallard is that:
- The three DIO0, DIO1 and DIO2 pins are multiplexed. If you do that you have to use diodes. See documentation.
- MISO, MOSI and CLK are hardwired and defined by the Wemos D1 mini board. That is why they cannot be changed.

The first time that the gateway comes up the filesystem is indeed formatted. Nothing to worry about. After the gateway is working for some time, the filesystem is formatted and the gwayConfig.txt file will be made.

The Gateway ID and registration is not necessary for receiving messages, but indeed you are advised to register the gateway with TTN )some of mine are registered and other not).

Maarten

closed 2018-06-12 by platenspeler