things4u/ESP-1ch-Gateway-v5.0--OLD #24: Time Bug & Don't work WIFI manager

VladimirBakum, 2018-05-24:
In the WEB manager I see:

> Version: V.5.1.1.H; 180517a
ESP alive since Thursday 24-5-2018 19:57:20, Uptime: 0-00:56:09
Current time Thursday 7-2-2036 10:24:11

And wrong time on the OLED display (Time of packet receiving).

Also dont't work WIFI manager.

VladimirBakum, 2018-05-25:
Time and Date are normalized after ~ 22 hours of work.
Current Time & Date in WEB manager are normal now. On OLED screen too.
Strange bug. Sorry if I disturb you.... Maybe this bug manifesting only for me?
After power off and then power on, date & time is still normal.
I think that this issue can be closed.

platenspeler, 2018-06-02:
Thank you. Please keep an eye on it, as time synchronization is difficult and sometimes does not succeed for some time (does not seem to be a bug but just waits too long).

If you get more behviour like this please tell me.

Maarten

closed 2018-06-02 by platenspeler