things4u/ESP-1ch-Gateway-v5.0--OLD #30: Unknown transceiver=FF, pins.rst =0, pins.ss  =16, pins.dio0 =15, pins.dio1 =15, pins.dio2 =15

HWittlich, 2018-06-11:
I'm using a Wemos D1 Mini board and a hallard type board. 
Often, the abort is called after the WWW Server is started because:
```
Unknown transceiver=FF, pins.rst =0, pins.ss  =16, pins.dio0 =15, pins.dio1 =15, pins.dio2 =15
```
 but at other times I get a lot of 
`
FLAG  ::I=  --  , F=0, SF=7, E=0, S=RX   , eT=8763223, dT=8764264
FLAG  ::I=  --  , F=0, SF=7, E=0, S=RX   , eT=8769035, dT=8770076`
on the Serial Monitor. 

What am I doing wrong? Any ideas? Your help is greatly appreciated!

platenspeler, 2018-06-12:
I have released a new version of the software: 5.2.1 if all OK these messages should have disappeared by now. Please upgrade to the new version and tell me if these message keep popping up on the Serial port.

Okamensky, 2019-11-16:
Hello there, I have been trying to make a one channel gateway from ESP32 and RFM95W...

I managed to upload the sketch, but in Serial monitor I get a connection to wifi and then this:

Unknown transceiver=0, pins.rst =27, pins.ss  =16, pins.dio0 =26, pins.dio1 =33, pins.dio2 =32

abort() was called at PC 0x400db806 on core 1

I tried version 5.3.3 as well as 6.1.1 and still no luck