things4u/ESP-1ch-Gateway-v5.0--OLD #33: Heltec WiFi LoRa 32 Oled problem

webfrank, 2018-06-22:
Hi, I got an Heltec ESP32 with OLED (SSD1306) were I put the latest code properly configured with OLED 1 but nothing is displayed.

I changed oLED.h file to define OLED_SCL and OLED_SDA with PINs 15 and 4 respectively for _PIN_OUT==5 (which I use) but display remains black.

Any suggestion?

platenspeler, 2018-06-23:
Hi,

I do have it working for the 433MHz version of the TTGO/Heltec board, and indeed I made changes to the I2C pins like you did.

Will publish shortly, but still testing some changes I made for the WiFi, I do not remember excatly but there were issues. I init the board with the OLED_RST==16 pin before getting it to work.

So add these lines at the beginning of the init_oLED() function in _OLED.ino:

#if defined OLED_RST
	pinMode(OLED_RST,OUTPUT);
	digitalWrite(OLED_RST, LOW); // low to reset OLED
	delay(50); 
	digitalWrite(OLED_RST, HIGH); // must be high to turn on OLED
	delay(50);
#endif

platenspeler, 2018-06-23:
And 

#define OLED_RST 16

Somewhere in the OLED.h file in the right section

webfrank, 2018-06-24:
Hi, it worked, will you do a PR?

platenspeler, 2018-06-24:
Ok, thanks. I will publish a new version of the code shortly. Still finishing tests for getting a gateway working on 433.175 mhz an 433.375 and 433.575 as well as some new WiFi functions and better support for ttgo/heltec modules.

platenspeler, 2018-06-28:
Done

closed 2018-06-28 by platenspeler