things4u/ESP-1ch-Gateway-v5.0--OLD #36: compiler issue

skyladle, 2018-07-05:
hardware : TTGO 2.0
compiler : Arduino IDE 1.8.3
broad: ESP32 Dev Module
however a compiler error , can help me tell me what missing. sorry a beginner question

C:\Users\Myfamily\Downloads\arduino_ttgo_lora\ESP-1ch-Gateway-v5.0-master (2)\ESP-1ch-Gateway-v5.0-master\ESP-sc-gway\_txRx.ino: In function 'int sendPacket(uint8_t*, uint8_t)':

_txRx:123: error: 'base64_dec_len' was not declared in this scope

  uint8_t payLength = base64_dec_len((char *) data, strlen(data));

                                                                ^

_txRx:126: error: 'base64_decode' was not declared in this scope

  base64_decode((char *) payLoad, (char *) data, strlen(data));

                                                             ^

C:\Users\Myfamily\Downloads\arduino_ttgo_lora\ESP-1ch-Gateway-v5.0-master (2)\ESP-1ch-Gateway-v5.0-master\ESP-sc-gway\_txRx.ino: In function 'int buildPacket(uint32_t, uint8_t*, LoraUp, bool)':

_txRx:381: error: 'base64_enc_len' was not declared in this scope

  int encodedLen = base64_enc_len(messageLength);  // max 341

                                               ^

_txRx:388: error: 'base64_encode' was not declared in this scope

  base64_encode(b64, (char *) message, messageLength);// max 341

                                                    ^

Multiple libraries were found for "WiFiUdp.h"
 Used: D:\arduino\1.8.3\arduino-1.8.3-windows\arduino-1.8.3\hardware\espressif\esp32\libraries\WiFi

platenspeler, 2018-07-07:
Did you specify the location of the sketch in the IDE in <File><Preferences> under sketchbook location? And did you include all the libraries that were included with the sketch (such as base64?)

Maarten

skyladle, 2018-07-08:
@Maarten ,thank , i put it at wrong location.it solved.

closed 2018-07-08 by skyladle