things4u/ESP-1ch-Gateway-v5.0--OLD #39: inDecodes functions Doesn't work

AI2M, 2018-07-10:
I think, this function need more parameters.

platenspeler, 2018-07-10:
Can you please explain? Doesn't work?

AI2M, 2018-07-11:
int inDecodes(char * id) {

	uint32_t ident = ((id[3]<<24) | (id[2]<<16) | (id[1]<<8) | id[0]);

	int i;
	for ( i=0; i< (sizeof(decodes)/sizeof(codex)); i++) {
		if (ident == decodes[i].id) {
			return(i);
		}
	}
	return(-1);
}

[image]
In _utils decodes and codex don't declared in function.

peterkorolev, 2018-07-11:
Comment out the function "inDecodes," and set TRUSTED_DECODE in ESP-sc-gway.h to 0.

AI2M, 2018-07-11:
To peterkorolev.
Yes, I solve this issue like yours suggestion. but i think it will be effect to other.
 it's seem works but  gateway can't send data to node.

platenspeler, 2018-07-11:
I do not think that sending data to the node it is related to your first issue. 

For the inDecodes function I have added

#if _LOCALSERVER==1 

It should only be defined and work when _LOCALSERVER is set. I have uploaded the change to github.

Maarten

AI2M, 2018-07-12:
To platenspeler.

Thank You, So i have a question. This example code can forward downlink to node, or not ?

platenspeler, 2018-07-12:
As far as I know the gateway does send downstream messages correctly. It is a little difficult for me to test as we have quite a few gateways around the house. I plan to test the downstream function on ESP32 again in August.

closed 2018-08-07 by platenspeler