things4u/ESP-1ch-Gateway-v5.0--OLD #40: Heltec ESP32 sends to TTN but does not get data

LucasSantosSilva, 2018-07-10:
The connection information even arrives at the TTN, but does not show the data. I am using the same code obtained here. What may be the problem?

[image]

I don't know if this part send the data, I tried with values fixed too.

`    void do_send(osjob_t* j)
{
  // Check if there is not a current TX/RX job running
  if (LMIC.opmode & OP_TXRXPEND) 
  {
    Serial.printf("OP_TXRXPEND, not sending\r\n");
  } 
  else
  if (!(LMIC.opmode & OP_TXRXPEND)) 
  {
    TempAndHumidity newValues = dht.getTempAndHumidity();
    lpp.reset();
    newValues.temperature = 11;
    newValues.humidity = 40;
    lpp.addTemperature(1, newValues.temperature);
    lpp.addRelativeHumidity(2, newValues.humidity);
    
    Serial.printf("Temperature : %.2f, Humidity : %.2f\r\n", newValues.temperature, newValues.humidity);
    display.clear();
    display.drawString(0, 0, "Temperature: ");
    display.drawString(90, 0, String(newValues.temperature));
    display.drawString(0, 20, "Humidity  : ");
    display.drawString(90,20, String(newValues.humidity));
    display.display();    

#if 0    
    if (GPS.fix)
    {
      float lat = GPS.latitudeDegrees;
      float lon = GPS.longitudeDegrees;
      float alt = GPS.geoidheight;
      Serial.printf("GPS lat = %.2f, lon = %.2f, alt = %.2f\n", lat, lon, alt);
      lpp.addGPS(3, lat, lon, alt);
    }
#endif
    LMIC_setTxData2(1, lpp.getBuffer(), lpp.getSize(), 0);   
    Serial.printf("Packet queued\r\n");
  }
}   `

platenspeler, 2018-07-10:
Hi,
I do not quite know how we can help you. This page is for supporting the single channel gateway but you have a question regarding an ESP32 sensor node. Do you use the single channel gateway or another gateway?
If the single channel gateway is used, it is possible to see the messages received and even decode them (#define _LOCALSERVER 1).
But if you are not using a single channel gateway you better ask on the TTN forum for ESP32 nodes.

LucasSantosSilva, 2018-07-10:
Yes, I using the single channel getaway in my WiFi LoRa 32, I will try on this part #LOCALSERVER1

 if here is not this local appropriate I apologize

platenspeler, 2018-07-10:
#define _LOCALSERVER 1
see ESP-sc-gway.h and fill in your sensor in sensor.h

LucasSantosSilva, 2018-07-12:
I solved the problem, it was not a code, but a TTN configuration. Thanks =)

closed 2018-07-12 by LucasSantosSilva