things4u/ESP-1ch-Gateway-v5.0--OLD #41: Errors during compilation for ESP32

DooMMasteR, 2018-07-13:
I tried to compile the gw software for ESP32 and did everything according to the guide, but somehow my WiFiClient.h in the ESP32-libs seems to be messed up or is not found at all? 

```
Using board 'lolin32' from platform in folder: /home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32
Using core 'esp32' from platform in folder: /home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32
Build options changed, rebuilding all
Detecting libraries used...
"/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/config" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/bluedroid/api" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/app_trace" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/bootloader_support" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/bt" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/driver" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/esp_adc_cal" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/esp_http_client" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/esp-tls" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/heap" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/jsmn" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/log" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/mdns" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/mbedtls_port" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/sdmmc" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/smartconfig_ack" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/spiffs" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/ulp" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/wear_levelling" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/coap" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/console" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/expat" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/json" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/lwip" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/nghttp" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/soc" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/wpa_supplicant" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10802 -DARDUINO_LOLIN32 -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="LOLIN32" -DARDUINO_VARIANT="lolin32"  -DESP32 -DCORE_DEBUG_LEVEL=0  "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/cores/esp32" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/variants/lolin32" "/tmp/arduino_build_706134/sketch/ESP-sc-gway.ino.cpp" -o "/dev/null"
"/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/config" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/bluedroid/api" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/app_trace" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/bootloader_support" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/bt" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/driver" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/esp_adc_cal" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/esp_http_client" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/esp-tls" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/heap" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/jsmn" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/log" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/mdns" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/mbedtls_port" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/sdmmc" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/smartconfig_ack" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/spiffs" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/tools/sdk/include/ulp" "-I/home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espre [... cut ...]
Using library FS at version 1.0 in folder: /home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/libraries/FS 
Using library gBase64 at version 1.0.0 in folder: /home/arntz/build/ESP-1ch-Gateway-v5.0/libraries/gBase64 
Using library ESP8266_Oled_Driver_for_SSD1306_display at version 3.2.7 in folder: /home/arntz/build/ESP-1ch-Gateway-v5.0/libraries/ESP8266_Oled_Driver_for_SSD1306_display 
Using library Wire at version 1.0 in folder: /home/arntz/build/ESP-1ch-Gateway-v5.0/hardware/espressif/esp32/libraries/Wire 
exit status 1
Error compiling for board WEMOS LOLIN32.
```

DooMMasteR, 2018-07-13:
Simply removing the line, at least get's it compiled.

per1234, 2018-07-13:
The problem is caused by the incorrect filename case (WiF**I**Client.h vs. WiF**i**Client.h). This will only occur on a filename case sensitive OS like Linux.

DooMMasteR, 2018-07-13:
Correct.
But since the included is not needed anyway, I simply dropped it.

platenspeler, 2018-08-07:
Ik close this issue as the code had been updates.

closed 2018-08-07 by platenspeler