things4u/ESP-1ch-Gateway-v5.0--OLD #43: Using SPIFFS on LoRaGO Dock

trlafleur, 2018-07-13:
Using 5.3.2, cant's seem to get file system to work... any changes needed for the LoRaGO Dock ver 1.1 board?? Am I missing something??

When I try to format from the web page, it tries as you can see below...

~~~
load 0x4010f000, len 1384, room 16 
Assert=Do Asserts
debug=1
readConfig:: Starting 
ERROR:: readConfig, file=/gwayConfig.txt does not exist .. FormattingMAC: 18:fe:34:fa:99:3d, len=17
WlanConnect:: Init para 0
0:1. WiFi connect SSID=lafleur, pass=sonyxxxxx
WlanCStatus:: CONNECTED tolafleur
WARNING:: writeConfig, file not exists, formatting /gwayConfig.txt
ERROR:: writeConfig, open file=/gwayConfig.txt
Host esp8266-fa993d WiFi Connected to lafleur on IP=192.168.20.226
Local UDP port=1700
Connection successful
Gateway ID: 18FE34FFFFFB993D, Listening at SF8 on 903.90 Mhz.
setupOta:: Started
Ready
IP address: 192.168.20.226
Time: Friday 09:36:59
WARNING:: writeConfig, file not exists, formatting /gwayConfig.txt
ERROR:: writeConfig, open file=/gwayConfig.txt
Gateway configuration saved
WWW Server started on port 80
WARNING:: writeConfig, file not exists, formatting /gwayConfig.txt
ERROR:: writeConfig, open file=/gwayConfig.txt
--------------------------------------
ERROR:: addLog:: file=/log-0 does not exist .. rec=1
file open failed=
/log-0
PKT_PULL_RESP:: received
sendPacket:: LoraDown filled
sendPacket:: fini OK
PKT_TX_ACK:: tmst=64947142
PKT_PULL_RESP:: size 197 From 13.66.213.36, port 1700, data: txpk...
ERROR:: addLog:: file=/log-0 does not exist .. rec=2
file open failed=
/log-0
PKT_PULL_RESP:: received
sendPacket:: LoraDown filled
sendPacket:: fini OK
PKT_TX_ACK:: tmst=118709560
PKT_PULL_RESP:: size 198 From 13.66.213.36, port 1700, data: txpk...
ERROR:: addLog:: file=/log-0 does not exist .. rec=3
file open failed=
/log-0
PKT_PULL_RESP:: received
sendPacket:: LoraDown filled
sendPacket:: fini OK
PKT_TX_ACK:: tmst=716422410
PKT_PULL_RESP:: size 198 From 13.66.213.36, port 1700, data: txpk...
FORMAT ...DONE
FORMAT ...DONE
FORMAT ...DONE
FORMAT ...DONE
FORMAT ...DONE
FORMAT ...DONE
FORMAT ...DONE
FORMAT ...DONE
FORMAT ...DONE
FORMAT ...DONE
FORMAT ...DONE
FORMAT ...DONE
FORMAT ...DONE
FORMAT ...DONE
FORMAT ...DONE
FORMAT ...DONE
FORMAT ...DONE
WARNING:: writeConfig, file not exists, formatting /gwayConfig.txt
ERROR:: writeConfig, open file=/gwayConfig.txt

~~~

platenspeler, 2018-07-16:
Hmm, occasionaly the filesystem gets corrupted. Especially when pulling the usb plug during file operationa. The solution is to reformat the filesystem again (sorry) by defining FORMAT in the .h file by 1. Do not forget to reset to 0 after succesful format.

platenspeler, 2018-07-16:
Btw. I do not recognize that situation. Will have a look.

platenspeler, 2018-07-17:
What board type did you use and what pin configuration in ESP-sc-gway.h
And does the "old" version of the software as used by SandboxElectronics work?

I do not have one myself so I cannot try.

Maarten

fewi, 2018-07-18:
I got my LoraGo Board today, flashed the newest version and can confirm the errors. 

Used: #define _PIN_OUT 1 

Tomorrow I will try to format the filesystem and check if the error still exists.

fewi, 2018-07-18:
I also noticed that the gateay always starts with SF8 but I set it in .h file to SF7. 
`#define _SPREADING SF7`

> Connection successful
Gateway ID: 2C3AE8FFFF45106E, Listening at SF8 on 868.10 Mhz.
setupOta:: Started
Ready
IP address: 192.168.0.52
Time: Wednesday 16:41:16
WARNING:: writeConfig, file not exists, formatting /gwayConfig.txt
ERROR:: writeConfig, open file=/gwayConfig.txt
Gateway configuration saved
WWW Server started on port 80

Dont know if this relates to the same problem or if I should open a new issue.

platenspeler, 2018-07-18:
Pin OUt 1 seems to be correct looking at the SandBoxElectronics version.
They seems to use the multiplexing function for pin DIO0, DIO1 and DIO2 just like the Hallard board does. Let me know whether this version works.

Maarten

platenspeler, 2018-07-18:
W.r.t. spreading factor. A bug in _loraFIles.ino on line 47. I will set it to _SPREADING in the next version.

Maarten

trlafleur, 2018-07-18:
I'm using ver 1.1 of the Sandbox board.... It did have ver 4.xx on it when
I receive it, but I was having issues recompiling the code, so I went
directly to your current version...  If I have some time I will try to get
the 4.xx version up again...

I also noted the SF8 issues...

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

fewi, 2018-07-19:
Tryed to fomat filesystem but didnt solve the issue.

Then I tryed to flash v4.x and the Sandbox version. In both versions I get a compile error also mentioned in this issue: https://github.com/things4u/ESP-1ch-Gateway-v4.0/issues/26

trlafleur, 2018-09-16:
Looks like this is now fixed in 5.3.3

closed 2019-11-04 by platenspeler