things4u/ESP-1ch-Gateway-v5.0--OLD #47: error massage during compiling: 'ip_addr_t' does not name a type  ip_addr_t      dns_getserver(u8_t numdns);

burkadius, 2018-08-07:
Arduino: 1.8.5 (Mac OS X), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

Sketch wird kompiliert...

In file included from /Users/marlisdrechsler/Documents/Arduino/ESP-sc-gway/ESP-sc-gway.ino:63:0:
/Users/marlisdrechsler/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc2/tools/sdk/lwip/include/lwip/dns.h:104:54: error: 'ip_addr_t' has not been declared
 typedef void (*dns_found_callback)(const char *name, ip_addr_t *ipaddr, void *callback_arg);
                                                                                               ^
/Users/marlisdrechsler/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc2/tools/sdk/lwip/include/lwip/dns.h:108:43: error: 'ip_addr_t' has not been declared
 void           dns_setserver(u8_t numdns, ip_addr_t *dnsserver);
                                                                    ^
/Users/marlisdrechsler/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc2/tools/sdk/lwip/include/lwip/dns.h:109:1: error: 'ip_addr_t' does not name a type
 ip_addr_t      dns_getserver(u8_t numdns);
 ^
/Users/marlisdrechsler/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc2/tools/sdk/lwip/include/lwip/dns.h:110:56: error: 'ip_addr_t' has not been declared
 err_t          dns_gethostbyname(const char *hostname, ip_addr_t *addr,
                                                                                              ^

Fehler beim Kompilieren für das Board WeMos D1 R2 & mini.

burkadius, 2018-08-07:
Hi, can anybody help me to solve  this problem? When I had compiled and installed the version 5.3.1 every think was fine but by the latest version 5.3.2 this problem occurs .I nothing have change, the same notebook, IDE etc.
Thanks 
Burkhard

platenspeler, 2018-08-08:
Hi, 

Did you specify <Preferences><Sketchbook Location> well and did you include all libraries?
ip-addr-t is not used in the gateway code, so only in DNS library. It seems the library cannot find the definitions.

Maarten

burkadius, 2018-08-08:
ok, I will check it. Thank You

closed 2018-08-08 by burkadius

outthereandaway, 2019-02-04:
Hi,

I get the exact same error... I have all the libraries installed, I believe. How did you solve this issue? I just can't get it to compile... Thanks a lot!

I am using a LOLIN D1 mini.

/Users/nameofuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include/lwip/dns.h:109:1: error: 'ip_addr_t' does not name a type
 ip_addr_t      dns_getserver(u8_t numdns);
 ^
/Users/nameofuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include/lwip/dns.h:110:56: error: 'ip_addr_t' has not been declared
 err_t          dns_gethostbyname(const char *hostname, ip_addr_t *addr,
                                                        ^
Multiple libraries were found for "WiFiUdp.h"
 Used: /Users/nameofuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
Error compiling for board WeMos D1 R2 & mini.

VladimirBakum, 2019-02-04:
**error: 'ip_addr_t' has not been declared**

I had this issue many months ago. I thought that this problem was solved.
You need add one "include" line in ESP-sc-gway.ino like:

`extern "C" {
#include "user_interface.h" // !!! add in v5.2 - v5.3.3
#include "lwip/err.h"
#include "lwip/dns.h"
}
`
Versions 5.2; 5.3.1; 5.3.2 and 5.3.3 has this issue.
I hope this solve is help you.
With best wishes!

outthereandaway, 2019-02-04:
Hey @VladimirBakum 

It works!!! Thanks a lot!!! Problem solved.

Can't get WiFiManager.h to work... no idea why. But no intention to dig into that. Can live without wifimanager ;-)

All the best, cheers