things4u/ESP-1ch-Gateway-v5.0--OLD #51: ESP32 instruction

arrowcircle, 2018-10-23:
Hey!
I am trying to build single channel gateway, but cant compile project. It requires ESP8266WiFi.h in WiFi manager, but if ESP32 board is loaded it will not load esp8266 libs.
Any ideas how to compile it?

JackGruber, 2018-10-24:
Hi, 

have you point your Arduino Sketchbook path to your unziped ESP-1ch-Gateway-v5.0 folder with the containing libraries folder?

arrowcircle, 2018-10-24:
@JackGruber I figured out how to compile it, but there are changed in the code that needed to be done to compile successfully. As I remember, its about webserver and some libs.

gabrielklein, 2018-11-01:
https://github.com/kersing/ESP-1ch-Gateway-v5.0 but it's an older version.

gutembergsa, 2019-10-15:
> @JackGruber I figured out how to compile it, but there are changed in the code that needed to be done to compile successfully. As I remember, its about webserver and some libs.

can you explain what "changes"? having same issue here

platenspeler, 2019-11-17:
Is there something I must fix, or can you compile now?
I know that in version 6 most issues are fixed, but be sure to include the right version 6 of ArduinoJson.

platenspeler, 2019-11-28:
I will close this issue.

closed 2019-11-28 by platenspeler