things4u/ESP-1ch-Gateway-v5.0--OLD #52: No downlink

jpunn, 2018-12-02:
I tried to install ESP -Gateway -v5.0 on Esp32 TTGO board. you can see the output below.

 I have a problem with Rx window and downlink is not enable pleae see the pic1.
 this ttn-abp node is not listing in the gateway . please share the  github link to the node program which works with ESP -Gateway -v5.0 .

Please see the gateway output and TTN-node out. This TTN node is not communication with my gateway. The values shown in the log is from another node but there is no downlink. 
I am looking for a node which can connect to this gateway. 

[image]
[image]

VladoPortos, 2019-05-24:
Same issue I have, client will send packets but waits for confirmation that never arrives... no idea what to do about it.

bazfp, 2019-05-28:
https://github.com/things4u/ESP-1ch-Gateway-v5.0/pull/72

The JSON casting fix worked for me