things4u/ESP-1ch-Gateway-v5.0--OLD #55: Gatway hangs irremediably after some time...

alexbn71, 2019-01-13:
I've tried this software for days with two kind of hardware, one "SparkFun LoRa Gateway - 1-Channel (ESP32)" and one "TTGO LORA32" and the result is the same... after few hours or at most 1 or 2 days the gateways stop running. The web server does not repond... the serial consol does not show anything, like a reset or any other error.

Can you suggest me how to set properly the debug variables to try to catch what's going wrong?

fabioetec, 2019-03-08:
I have the same problem, did you find any solution?

Thank you
Fabio

nicolasimeoni, 2019-03-09:
see this pull request: 
https://github.com/things4u/ESP-1ch-Gateway-v5.0/pull/61
it fixes this problem
Nicola

platenspeler, 2019-11-17:
I did merge the lramodem.ini file. Seems this issue is resolved.

closed 2019-11-17 by platenspeler