things4u/ESP-1ch-Gateway-v5.0--OLD #58: TTGO ESP32 v2.1 1.6 feedback - pin-layout

matthys70, 2019-01-27:
I just want to address two issue with TTGO ESP32 V2.1 1.6, mind this is hardware specific.

I noticed there is a _PIN_OUT==5 part in loraModem.h, but needed to change this to:
```
#elif _PIN_OUT==5
// ----------------------------------------------------------------------------
// For ESP32/TTGO based board for EU32 with 0.9" OLED
// NOTE: This board shoudl be same as general type TTGO (nr 4)
// but for the moment we include this as a separate item
//
// SCK  == GPIO5/ PIN5
// SS   == GPIO18/PIN18 CS
// MISO == GPIO19/ PIN19
// MOSI == GPIO27/ PIN27
// RST  == GPIO23/ PIN23
struct pins {
	uint8_t dio0=26;		// GPIO26 / Dio0 used for one frequency and one SF
	uint8_t dio1=33;		// GPIO33 / Used for CAD, may or not be shared with DIO0
	uint8_t dio2=32;		// GPIO32 / Used for frequency hopping, don't care
	uint8_t ss=18;			// GPIO18 / Dx. Select pin connected to GPIO18
	uint8_t rst=23;			// GPIO23 / D3. Reset pin not used	
} pins;
#define SCK 5				// Check
#define MISO 19				// Check
#define MOSI 27				// Check
#define RST 23				// Check
#define SS 18
```

Also the OLED display was not working and wondering why there is no _PIN_OUT==5 part in oLED.h. So I added this (also different pin-layout):
```
#elif _PIN_OUT==5                // TTGO (onboard version used, also for DIY)
#define OLED_SCL 22               // GPIO22 / 
#define OLED_SDA 21               // GPI021 / 
```

I guess it would have been better to just create a hardware config file where you add all these pin-layouts etc.

Anyway, just some feedback if somebody else has/had the same issue.

sle118, 2019-04-04:
I did solve the oled issue on that same board by manually calling SPI init function, passing pins. This persists throughout the execution without having to play with include files

platenspeler, 2019-11-18:
If this is solved, I close the issue.\\Thankjs,
Maarten

closed 2019-11-18 by platenspeler