things4u/ESP-1ch-Gateway-v5.0--OLD #65: error compiling void printIP(IPAddress, const char*, String&)

joniyes, 2019-03-08:
Hi, i have problem with compiling project:

C:\Users\d3el\Documents\Arduino\ESP-sc-gway\_wwwServer.ino: In function 'void printIP(IPAddress, const char*, String&)':
_wwwServer:45:10: error: ambiguous overload for 'operator+=' (operand types are 'String' and 'IPAddress')
  response+=(IPAddress)ipa[0]; response+=sep;
          ^
C:\Users\d3el\Documents\Arduino\ESP-sc-gway\_wwwServer.ino:45:10: note: candidates are:
In file included from C:\Users\d3el\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/Arduino.h:261:0,
                 from sketch\ESP-sc-gway.ino.cpp:1:
C:\Users\d3el\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/WString.h:120:18: note: String& String::operator+=(const char*) <near match>
         String & operator +=(const char *cstr) {
                  ^
C:\Users\d3el\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/WString.h:120:18: note:   no known conversion for argument 1 from 'IPAddress' to 'const char*'

C:\Users\d3el\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/WString.h:124:18: note: String& String::operator+=(char)
         String & operator +=(char c) {
                  ^
C:\Users\d3el\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/WString.h:128:18: note: String& String::operator+=(unsigned char)
         String & operator +=(unsigned char num) {
                  ^
C:\Users\d3el\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/WString.h:132:18: note: String& String::operator+=(int)
         String & operator +=(int num) {
                  ^
C:\Users\d3el\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/WString.h:136:18: note: String& String::operator+=(unsigned int)
         String & operator +=(unsigned int num) {
                  ^
C:\Users\d3el\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/WString.h:140:18: note: String& String::operator+=(long int)
         String & operator +=(long num) {
                  ^
C:\Users\d3el\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/WString.h:144:18: note: String& String::operator+=(long unsigned int)
         String & operator +=(unsigned long num) {
                  ^
C:\Users\d3el\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/WString.h:148:18: note: String& String::operator+=(float)
         String & operator +=(float num) {
                  ^
C:\Users\d3el\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/WString.h:152:18: note: String& String::operator+=(double)
         String & operator +=(double num) {
                  ^
C:\Users\d3el\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/WString.h:156:18: note: String& String::operator+=(const __FlashStringHelper*) <near match>
         String & operator += (const __FlashStringHelper *str){
                  ^

refe, 2019-03-19:
in Arduino IDE , go to Board Managers, search esp , select esp8266 by ESP8266 Community and use version 2.5.0-beta2

platenspeler, 2019-11-17:
In _wwwServer.ino the following needs to be changed:
// ----------------------------------------------------------------------------
// PRINT IP
// Output the 4-byte IP address for easy printing.
// As this function is also used by _otaServer.ino do not put in #define
// ----------------------------------------------------------------------------
static void printIP(IPAddress ipa, const char sep, String& response)
{
	response+=(String)ipa[0]; response+=sep;
	response+=(String)ipa[1]; response+=sep;
	response+=(String)ipa[2]; response+=sep;
	response+=(String)ipa[3];
}

closed 2019-11-17 by platenspeler