things4u/ESP-1ch-Gateway-v5.0--OLD #70: Compilation problems about _txRx

richard-ubi, 2019-04-03:
_txRx:63:2: error: 'StaticJsonDocument' was not declared in this scope
StaticJsonDocument<312> jsonBuffer;

gfnord, 2019-04-03:
I'm getting the same issue here.

gfnord, 2019-04-03:
You just need to install/update the ArduinoJson lib to version 6.10

Option 1: Use the Arduino Library Manager

Difficulty: ⚫⚪⚪

Recommendation: Use this method if you use Arduino 1.6.x or newer.

Instructions:

Open the Arduino Library Manager
Search for “ArduinoJson”
Select the version: 6.10.0 
Click install.

richard-ubi, 2019-04-04:
> You just need to install/update the ArduinoJson lib to version 6.10
> 
> Option 1: Use the Arduino Library Manager
> 
> Difficulty: ⚫⚪⚪
> 
> Recommendation: Use this method if you use Arduino 1.6.x or newer.
> 
> Instructions:
> 
> Open the Arduino Library Manager
> Search for “ArduinoJson”
> Select the version: 6.10.0
> Click install.
> 
> After updating here I could compile without any problem.

Thank you very much. I have solved the problem

gfnord, 2019-04-04:
Okay, you may close this issue, please.

VladimirBakum, 2019-04-04:
Today I had the same issue but this method is don't work. At home compiling ok without update version (Win7x64), but at work I have the same error (Win10x64).

VladimirBakum, 2019-04-05:
I changed the version of **Arduino** from 1.8.9 (Windows Store 1.8.21.0) to 1.8.5 - now compiling without any errors. Win10x64. 
Note: ArduinoJson by Benoit Blanchon version 5.13.1

platenspeler, 2019-04-07:
Glad to hear that the new library works. I will close the issue in 2 weeks.

closed 2019-04-08 by richard-ubi

mrx23dot, 2019-05-10:
Please update main read me (show old version)

richard-ubi, 2019-05-11:
ok ! i am so honored for your reply !i had tested ,but i find that it has serious packet loss rate while the different sf ,how do you think that

richard@ubitech.hk
 
From: Kis Gabor
Date: 2019-05-11 04:11
To: things4u/ESP-1ch-Gateway-v5.0
CC: richard-ubi; State change
Subject: Re: [things4u/ESP-1ch-Gateway-v5.0] Compilation problems about _txRx (#70)
Please update main read me (show old version)