things4u/ESP-1ch-Gateway-v5.0--OLD #76: Guru Meditation Error when ACK is required

petrkr, 2019-04-20:
When I send packet which have enabled ACK. GW will reboot itself with panic error

ESP32, latest master

SF12, channel 0

```
--------------------------------------
M readUdp:: PKT_PULL_RESP received
A readUdp:: Error: PKT_PULL_RESP sendPacket failed
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d3eaa  PS      : 0x00060a30  A0      : 0x800db0fd  A1      : 0x3ffb1f00  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffc1dfc  A5      : 0x3ffbeca3  
A6      : 0x00000003  A7      : 0x00060423  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000022  A11     : 0x00000000  A12     : 0x3ffc1dfc  A13     : 0x3ffbeca3  
A14     : 0x00000000  A15     : 0x3ffb0060  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d3eaa:0x3ffb1f00 0x400db0fa:0x3ffb1f20 0x400de49b:0x3ffb1f50 0x400de724:0x3ffb1f90 0x400ea495:0x3ffb1fb0 0x40088f51:0x3ffb1fd0
```

petrkr, 2019-04-20:
Now I recognized it's also when Ack is not required... But I got stacktrace:

```
PC: 0x400d3eaa: sendPkt(unsigned char*, unsigned char) at /home/petrkr/Arduino/ESP-1ch-Gateway-v5.0/ESP-sc-gway/_loraModem.ino line 603
EXCVADDR: 0x00000000

Decoding stack results
0x400d3eaa: sendPkt(unsigned char*, unsigned char) at /home/petrkr/Arduino/ESP-1ch-Gateway-v5.0/ESP-sc-gway/_loraModem.ino line 603
0x400db0fa: txLoraModem(unsigned char*, unsigned char, unsigned int, unsigned char, unsigned char, unsigned int, unsigned char, unsigned char) at /home/petrkr/Arduino/ESP-1ch-Gateway-v5.0/ESP-sc-gway/_loraModem.ino line 774
0x400de49b: stateMachine() at /home/petrkr/Arduino/ESP-1ch-Gateway-v5.0/ESP-sc-gway/_stateMachine.ino line 812
0x400de724: loop() at /home/petrkr/Arduino/ESP-1ch-Gateway-v5.0/ESP-sc-gway/ESP-sc-gway.ino line 1294
0x400ea495: loopTask(void*) at /home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.2/cores/esp32/main.cpp line 19
0x40088f51: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143```

petrkr, 2019-04-21:
This one solved by fix downlink messages

```
--- a/ESP-sc-gway/_txRx.ino
+++ b/ESP-sc-gway/_txRx.ino
@@ -94,7 +94,7 @@ int sendPacket(uint8_t *buf, uint8_t length)
        // {"txpk":{"codr":"4/5","data":"YCkEAgIABQABGmIwYX/kSn4Y","freq":868.1,"ipol":true,"modu":"LORA","powe":14,"rfch":0,"size":18,"tmst":1890991792,"datr":"SF7BW125"}}
 
        // Used in the protocol of Gateway:
-  JsonObject root = jsonBuffer.to<JsonObject>();
+  JsonObject root = jsonBuffer.as<JsonObject>();
```

but still downlink messages are not catched by host. Maybe some other problem with tester, have to check more... Atleast it does not longer panic

kami83, 2019-05-12:
Hi,

anything new to this? I have got the same problem.

Thanks a lot.

kami

VladoPortos, 2019-05-25:
Same issue here, might this be cause be firewall ? I mean the lora GW backend is on the wifi but the wifi router would not forward connection toLoraGW from outside ( from app server / lora server ) ... also how to handle 2 LorGW on one NW when firewall can forward port 1700 only to one specific IP ?

VladoPortos, 2019-05-25:
Nope its not firewall, I have enabled NAT to specific LoraGW on the network, had only one turned on... Send message, message is sent ok, it appears in LoraApp server ... LoraGW crash with error ( same as in this topic ) and no ACK is received. No idea what might be the issue.

VladoPortos, 2019-05-26:
Solved the crashes using one of the pull request builds, but the ACK still never arrive to sender, I have no idea now where the issue can be... I'll wait for proper LoraGW and see if its in this 1Ch build or somewhere else in the pipes

petrkr, 2019-05-26:
You can use RTL-SDR to measure if it transmits. For me it transmits, but node does not "hear" it. Now I have Rpi based fullGW and it works. So I wonder where is problem... Anyway that guru meditation error when downlink is permormed was solved by change "to" to "as" at JSON object.. Check my comment.

VladoPortos, 2019-05-27:
I'm waiting for custom PCB to connect my RAK831 to rPi and test it out... also ordered some build ( one box ) LoraAPs

kami83, 2019-05-29:
Hi,

i changed "to" to "as" and after this i get on TTN always the message that everything runs fine?

Cu kami

petrkr, 2021-08-12:
Already outdated repository, closing issue.

closed 2021-08-12 by petrkr