things4u/ESP-1ch-Gateway-v5.0--OLD #78: ArduinoJson compilation error

mrx23dot, 2019-05-10:
Head doesnt compile with mentioned ArduinoJson (version 5.13.1):

ESP-1ch-Gateway-v5.0-master\ESP-sc-gway\_txRx.ino: In function 'int sendPacket(uint8_t*, uint8_t)':

_txRx:63: error: 'StaticJsonDocument' was not declared in this scope
  StaticJsonDocument<312> jsonBuffer;

it compiles with ArduinoJson (version 6.10.1)

gppk, 2019-05-31:
+1 I also experienced this issue.

Arduino IDE Version - 1.8.5

**Steps to Fix:**

1. Download Zip from ArduinoJson Github - https://github.com/bblanchon/ArduinoJson
2. Remove existing ArduinoJson folder in Documents>Arduino>Libraries
3. Install ArduinoJson zip via arduino IDE.

Compilation successful.

This could be fixed by updating the ArduinoJson packaged in this Git, but I'm not sure if that would break other things.

offle, 2019-07-05:
Hi there, same here
When compiling with a new Version 6 of ArduinoJson I get the following error when receiving a package. It looks to me that the data cannot be extracted from the json and so it's null. 

rxPkt:: t=09:23:01, f=1, sf=7, a=D001BF9E, flags=50, addr=0, len=23
RXDONE in dT=60289: I=RXDONE HEADER , F=1, SF=7, E=0, S=RX   , eT=5373120, dT=81929
G addLog:: fileno=0, rec=1: 1 78 D8 0 A4 CF 12 FF FF 5 1F EC {"rxpk":[{"tmst":111133045,"chan":0,"rfch":0,"freq":868.099975,"stat":1,"modu":"LORA","datr":"lallalalaa","codr":"4/5","lsnr":9,"rssi":-43,"size":23,"data":"...heressomedatastuff..."}]}
T sendPacket:: ERROR: data is NULL
A readUdp:: Error: PKT_PULL_RESP sendPacket failed

blueish4, 2019-11-03:
This seems like a documentation error, #67 merged support for ArduinoJson 6 without changing the documentation or bundled library. I don't get any compilation errors using v6.13.0, but I haven't tested all of the functionality yet.

platenspeler, 2019-11-05:
Sorry,

Yes the documentation for version 5 is nto up to date. You should use ArduinoJson version 6!
I did update the documentation to make it in line with version 6 of the gateway.

Maarten

PS. I changed this already in version 6 of the software (see the separate github tree).

platenspeler, 2019-11-28:
I will close this issue. Please use ArduinoJson6 when possible.

closed 2019-11-28 by platenspeler