things4u/ESP-1ch-Gateway-v5.0--OLD #79: Serial monitor error

dd223, 2019-05-15:
Hello,

I'm working with a nodemcu esp8266 and a Rfm95 to make a 1Ch Gateway
I'm very new to this, first time reading c++ code but it's understandable. My problem is that when I compile, transfer everything is fine, I open my serial monitor, it connects to wifi fine but then it comes to this point and abort everything: 

_Unknown transceiver=FF, pins.rst =0, pins.ss  =16, pins.dio0 =20, pins.dio1 =19, pins.dio2 =0
Abort called_

I don't really understand what I've done wrong, I've went on the net to find a schematic of the nodes and did it this way : 
_struct pins {
  uint8_t dio0=20;  // GPIO15 / D8. For the Hallard board shared between DIO0/DIO1/DIO2
  uint8_t dio1=19;  // GPIO15 / D8. Used for CAD, may or not be shared with DIO0
  uint8_t dio2=0;  // GPIO15 / D8. Used for frequency hopping, don't care
  uint8_t ss=16;    // GPIO16 / D0. Select pin connected to GPIO16 / D0
  uint8_t rst=0;    // GPIO 0 / D3. Reset pin not used  
  // MISO 12 / D6
  // MOSI 13 / D7
  // CLK  14 / D5 
} pins;_
[image]

but maybe I'm wrong, maybe it's not about this, Please help me

Here is my full serial monitor:
[serial monitor.txt](https://github.com/things4u/ESP-1ch-Gateway-v5.0/files/3181859/serial.monitor.txt)

Thank you!

chinanzio, 2019-06-16:
Hello, i don't know if it is too late, but it maybe helps others .... i had the same problem, i solved it by changing the wiring ... i am using a different hardware, but it still helps ...
So, with a WeMos D1 v3.0.0 mini and a RFM95 i had to configure the pins as follow in order to make it work 

struct pins {
	uint8_t dio0=D8;		// GPIO26 / Dio0 used for one frequency and one SF
	uint8_t dio1=D1;		// GPIO26 / Used for CAD, may or not be shared with DIO0
	uint8_t dio2=D3;		// GPI2O6 / Used for frequency hopping, don't care
	uint8_t ss=D0;			// GPIO18 / Dx. Select pin connected to GPIO18
	uint8_t rst=D3;			// GPIO0  / D3. Reset pin not used	
} pins;

this is exactly my hardware
https://www.google.es/search?q=lolin+wemos+d1+mini&client=ubuntu&hs=OSm&tbm=isch&source=iu&ictx=1&fir=ex-unGBHSjT-8M%253A%252CBrarWE1jLjTPuM%252C_&vet=1&usg=AI4_-kSLrZHy6ZqrqnxzZt16fizRUWjo1A&sa=X&ved=2ahUKEwix7sD_r-3iAhWs1uAKHQLAAUwQ9QEwDHoECAQQBg#imgrc=ex-unGBHSjT-8M:

https://www.google.es/search?client=ubuntu&hs=VSm&tbm=isch&sa=1&ei=x-EFXei7HYHUUYq6quAI&q=rfm95&oq=rfm95&gs_l=img.3..0l10.32508.35923..36322...0.0..0.75.359.5......0....1..gws-wiz-img.......35i39j0i67.B_Nt7buR8bA#imgrc=kLbKO9p6tMY1MM:

Cheers

platenspeler, 2019-11-28:
I will close this issue. Please read the documentation for known setups.

closed 2019-11-28 by platenspeler