things4u/ESP-1ch-Gateway-v5.0--OLD #82: No connection to TTN with Wemos D1 Mini

TobiasBg, 2019-06-21:
Hi!

I've built pretty much exactly [this adapter board](http://things4u.github.io/UserGuide/One%20Channel%20Gateway/images/DSCN3047.JPG), with a Wemos D1 Mini and an RFM95W, using the ComResult pin mapping.

I flashed the Wemos with the current master branch, but had to use the Arduino IDE 1.8.8 to work around issue #77. I'm using the ESP8266 Arduino core version 2.5.2.
Everything is loading fine, I don't get errors in the Serial terminal and I can access the webserver GUI without problems.
However, the gateway on the TTN website (registered as a packet forwarder with the ID that is shown in the Serial terminal, never shows as "(green dot) connected" but only as "(gray dot) not connected".

However, if I flash the same code on an ESP32 dev board (with a slightly different pin mapping), everything is working nicely, the gateway shows as "connected" in the TTN console.

So, I assume that something is wrong right now with the ESP8266-specific code (the libraries?).

Does anybody have an idea on what I could try? Are there ways to get more debug information somewhere about why the connection of the Wemos to TTN is not being established?

chinanzio, 2019-07-11:
Hello, i have it working with a WeMos D1 mini ... try the wiring configuration i suggest in #79

platenspeler, 2019-11-17:
I hope this issue is resolved, I cannot see the adapter board anymore.
However, do not use the pin-out of the wrong board: It will compile but it will not work.
Please make sure you use exactly the right pin-out (which is the most impoirtant thing once the garteway compiles OK).

closed 2019-11-17 by platenspeler