things4u/ESP-1ch-Gateway-v5.0--OLD #89: reset after receing an uplink

OstertagM, 2019-08-05:
my gateway is receiving after receiving an uplink why?

OstertagM, 2019-08-05:
A readUdp:: NTP msg rcvd
G addLog:: fileno=0, rec=1: 1 4 B4 0 3C 71 BF FF FF FF 1B DC {"rxpk":[{"tmst":29878605,"chan":0,"rfch":0,"freq":868.099975,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":10,"rssi":-30,"size":28,"data":"QGoUASaAAAABE0IDngpw1AhSBvFPUeSgOqMMSw=="}]}
A readUdp:: Error: PKT_PULL_RESP sendPacket failed
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d316a  PS      : 0x00060a30  A0      : 0x800da30d  A1      : 0x3ffb1f00  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffc1cfc  A5      : 0x3ffbdc77  
A6      : 0x00000003  A7      : 0x00060423  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000022  A11     : 0x00000000  A12     : 0x3ffc1cfc  A13     : 0x3ffbdc77  
A14     : 0x00000000  A15     : 0x3ffb0060  SAR     : 0x0000000e  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x400d316a:0x3ffb1f00 0x400da30a:0x3ffb1f20 0x400dd1e7:0x3ffb1f50 0x400dd470:0x3ffb1f90 0x400e8e81:0x3ffb1fb0 0x4008f7bd:0x3ffb1fd0

Rebooting...

RaveGun, 2019-08-07:
Hi,

I have the same issue on my side too.
If the debug level = 2 then it is possible to see the error:
`
00:51:58.258 -> {"txpk":{"imme":false,"tmst":129525217,"freq":868.099975,"rfch":0,"powe":14,"modu":"LORA","datr":"SF7BW125","codr":"4/5","ipol":true,"size":33,"ncrc":true,"data":"IPW3aWpLJmqs+KyKJcFMcjmamHDHjXiRIVCka9fttDfz"}}
00:51:58.258 -> <T sendPacket:: ERROR: data is NULL
00:51:58.305 -> A readUdp:: Error: PKT_PULL_RESP sendPacket failed
00:51:58.305 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
00:51:58.305 -> Core 1 register dump:
00:51:58.305 -> PC      : 0x400d402a  PS      : 0x00060a30  A0      : 0x800db3d1  A1      : 0x3ffb1f00  
00:51:58.305 -> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffc1fe8  A5      : 0x3ffbeca7  
00:51:58.305 -> A6      : 0x00000003  A7      : 0x00060423  A8      : 0x00000000  A9      : 0x00000000  
00:51:58.305 -> A10     : 0x00000022  A11     : 0x00000000  A12     : 0x3ffc1fe8  A13     : 0x3ffbeca7  
00:51:58.352 -> A14     : 0x00000000  A15     : 0x3ffb0060  SAR     : 0x0000000e  EXCCAUSE: 0x0000001d  
00:51:58.352 -> EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
00:51:58.352 -> 
00:51:58.352 -> Backtrace: 0x400d402a:0x3ffb1f00 0x400db3ce:0x3ffb1f20 0x400de7a7:0x3ffb1f50 0x400dea30:0x3ffb1f90 0x400e8255:0x3ffb1fb0 0x40088ddd:0x3ffb1fd0
`

RaveGun, 2019-08-07:
Aha, now I understood the down-link pull request.

Have a look at this change:
https://github.com/things4u/ESP-1ch-Gateway-v5.0/pull/72/commits/f3f4f194a6e1059c15b6397fd8bfc8a32d631ac9#diff-0a7fba24a3dd3d3ea7934244ea958757

Now it is working and is allowing another TTGO flashed with the PAX counter to join and send some data.

Thank you.

OstertagM, 2019-08-08:
Dear RaveGun,
OK I understand it. But with GitHub and pull requests I don't know how it works and what I have to do to get the code into my code. Is there any automatics. Thanks

OstertagM, 2019-08-08:
Seems to be working with it now. But the pull request! Thanks RaveGun.

closed 2019-08-08 by OstertagM