things4u/ESP-1ch-Gateway-v5.0--OLD #94: ESP 32 keeps rebooting

dksmar, 2019-09-18:
Good day 

I managed to upload the code to TTGO Lora32 Oled v1. I had all the time errors with Wifi Manager and Webserer so I deactivated them.  I have put NTP server 2.europe.pool.ntp.org.

The problem now is that all the time says STARTING and from serila interface i saw that keeps rebooting. See here below the log from Serial.

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8740
load:0x40080400,len:5788
entry 0x4008069c
ESP32 defined, freq=868100000 EU868
ARDUINO_ARCH_ESP32 defined
SPIFFS init success
Assert=Do Asserts
debug=1
readConfig:: Starting
.SSID=  Home N
.PASS=  Lotus_Elise
.CH=    0
.SF=    9
.FCNT=  0
.DEBUG= 1
.PDEBUG=0
.CAD=0
.HOP=0
.NODE=  0
.BOOTS= 192
.RESETS=        0
.WIFIS= 192
.VIEWS= 0
.REFR=  0
.REENTS=        0
.NTPETIM=       0
.NTPERR=        0
.NTPS=  0
.FILEREC=       0
.FILENO=        0
.FILENUM=       0
.EXPERT=        0
#

MAC: 30:ae:a4:fe:a6:44, len=17
WlanConnect:: Init para 0
0:1:3. WiFi connect SSID=Home N, pass=Lotus_Elise
A WlanStatus:: CONNECTED to Home N
Host esp32-fea644 WiFi Connected to Home N on IP=10.24.85.51
Local UDP port=1700
Connection successful
Gateway ID: 30AEA4FFFFFEA644, Listening at SF9 on 868.10 Mhz.
Setup:: ERROR hostByName NTP
abort() was called at PC 0x400da3ca on core 1

Backtrace: 0x4008cc20:0x3ffb1ee0 0x4008ce51:0x3ffb1f00 0x400da3ca:0x3ffb1f20 0x400dbff5:0x3ffb1f40 0x400e6e7f:0x3ffb1fb0 0x40088ba1:0x3ffb1fd0

Rebooting...

```
has anyone expirienced similar problem?

dksmar, 2019-09-20:
Finally worked! there is something wrong with my local LAN DHCP. I have put as secondary option the SSID of my grandmother's house (1 floor down) and worked! Also when I put my mobile as hotspot with credentials of my house wifi works!

closed 2019-09-20 by dksmar

strech345, 2019-12-16:
For me it also reboot always (TTGO 1.6).
What does it mean that partition is missing? And is it a wifi problem? I read on other posts that i need to change some dhcp settings because of dns? If so what concret i've to do?

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
ESP32 defined, freq=868100000 EU868
ARDUINO_ARCH_ESP32 defined
E (641) SPIFFS: spiffs partition could not be found
Assert=Do Asserts
debug=1
readConfig:: Starting 
M ERR:: readConfig, file=/gwayConfig.txt does not exist .. Formatting
E (2482) SPIFFS: spiffs partition could not be found
MAC: XXXXXXXXXXXX, len=17
WlanConnect:: Init para 0
0:1:3. WiFi connect SSID=XXX, pass=XXXX
A WlanStatus:: CONNECTED to XXXX
WARNING:: writeConfig, file not exists, formatting E (11501) SPIFFS: spiffs partition could not be found
/gwayConfig.txt
ERROR:: writeConfig, open file=/gwayConfig.txt
Host esp32-4361c8 WiFi Connected to XXXX on IP=192.168.43.132
Local UDP port=1700
Connection successful
Gateway ID: XXXX, Listening at SF9 on 868.10 Mhz.
setupOta:: Started
Ready
IP address: 192.168.43.132
Time: Monday 02:33:54
WARNING:: writeConfig, file not exists, formatting E (13438) SPIFFS: spiffs partition could not be found
/gwayConfig.txt
ERROR:: writeConfig, open file=/gwayConfig.txt
Gateway configuration saved
WWW Server started on port 80
Unknown transceiver=0, pins.rst =0, pins.ss  =16, pins.dio0 =15, pins.dio1 =15, pins.dio2 =15

abort() was called at PC 0x400da6ca on core 1

Backtrace: 0x4008cc2c:0x3ffb1ec0 0x4008ce5d:0x3ffb1ee0 0x400da6ca:0x3ffb1f00 0x400daa06:0x3ffb1f20 0x400dc3c9:0x3ffb1f40 0x400e9be7:0x3ffb1fb0 0x40089341:0x3ffb1fd0

Rebooting...
```

strech345, 2019-12-16:
Ok 've solved by this function
https://github.com/things4u/ESP-1ch-Gateway-v6.0/issues/5