things4u/ESP-1ch-Gateway-v5.0--OLD #95: VScode platformIO compiler error

richard-ubi, 2019-10-14:
Hi friends！
VScode platformIO compile the following error, previously was successful, I hope that a friend can guide the correction method, thank you
`PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10004.191002 (1.0.4), tool-esptoolpy 1.20600.0 (2.6.0)
UnicodeEncodeError: 'gbk' codec can't encode character '\xbf' in position 40285: illegal multibyte sequence:
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Administrator\.platformio\platforms\espressif32\builder\main.py", line 209:
    target_elf = env.BuildProgram()
  File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 110:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 288:
    env.ConvertInoToCpp()
  File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 198:
    out_file = c.convert(ino_nodes)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 57:
    return self.process(contents)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 80:
    assert self._gcc_preprocess(contents, out_file)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 90:
    fp.write(contents)`

fmuntean, 2019-10-21:
having the same issue. found this: https://github.com/platformio/platform-espressif8266/issues/175

JackGruber, 2019-10-21:
Hi,

change the `platformio.ini`

From:
```
platform = espressif8266
````
to
```
platform = https://github.com/platformio/platform-espressif8266.git
```
this use the upstream version of espressif8266 and with this the error is gone.

richard-ubi, 2019-10-21:
OK, thx, I solved the problem by reinstalling and configuring python

closed 2019-10-21 by richard-ubi