things4u/ESP-1ch-Gateway-v5.0--OLD #98: Semtech SX1262

skyden-ua, 2020-02-11:
Hello. Your project helped me a lot, I continue to experiment with it, and these are the only nodes for hundreds of kilometers. I want to try the Semtech SX1262 and the E22-900M30S seems to be a good option for this.
http://www.ebyte.com/en/product-view-news.aspx?id=453

Could you add support for the Semtech SX1262?
Unfortunately, I do not have enough knowledge to do this.
Thank you very much for your work.

platenspeler, 2020-02-13:
Thanks!
As you do not have all the knowledge, do you know how to make a sensor?
I have ordered a board and will have a look to see if the code would change a lot.

M.

platenspeler, 2020-02-13:
Oh and btw i will work on version 6, the ESP 1 channel gateway version (without version 5)

skyden-ua, 2020-02-15:
> Благодарность!
> Поскольку у вас нет всех знаний, вы знаете, как сделать датчик?
> Я заказал плату и посмотрю, сильно ли изменится код.
> 
> М.

There are several ideas on how to do this. But the modules are still on the way and I'm not sure if this will work))

skyden-ua, 2020-03-31:
> Да, и между прочим, я буду работать над версией 6, 1-канальной шлюзовой версией ESP (без версии 5).

Hello. There are some successes with the boards E22-900M30S, Semtech SX1262