things4u/ESP-1ch-Gateway-v5.0--OLD #99: setup:: ERROR Time not set (yet).

cirujadigital, 2020-02-27:
I'm having some problems with the NTP server, I tried three different combinations, and I always have the same error:

WlanConnect:: Now starting WlanStatus
WlanStatus:: CONNECTED ssid=WayFay
WlanConnect:: WlanStatus Connected
MAC: ****************, len=17
Host=esp8266-3c8634 WiFi Connected to WayFay on IP=192.168.2.113
Local UDP port=1700
UDP Connection successful
Gateway ID:****************, Listening at SF7 on 903.90 MHz.
resolveHost:: OK=ar.pool.ntp.org IP=162.159.200.123
setup:: ERROR Time not set (yet). Time=-17829890
setup:: ERROR Time not set (yet). Time=-17829890

My setup is ESP8266+RFM95W, No GPS, No OLED and no gateway sensor.

I tried with different ntp servers, as listed for my country, but all with the same error:

ar.pool.ntp.org
http://1.south-america.pool.ntp.org/
http://0.south-america.pool.ntp.org/

platenspeler, 2020-02-29:
Please upgrade to the latest version of the software. Version 5 is there fore historical reasons, ad the ESP-1ch_Gateway (without the version number) is the latest version

regards,

Maarten

viniciusvdr01, 2022-04-12:
Was solved? I'm having the same problem and using the latest version.